RHSResearchLLC/NiteFury-and-LiteFury #8: Configuring LiteFury

lleon95, 2020-08-30:
Hi,

I am trying to configure the LiteFury with an FT232R connected to the JTAG port of the FPGA. I chose a 3.3V configuration and wiring the FT232R based on the following schematic: http://vak.ru/doku.php/proj/bitbang/bitbang-jtag and the JTAG port pinout from the Litefury [datasheet](https://github.com/RHSResearchLLC/NiteFury-and-LiteFury/blob/master/Hardware/uEVB.pdf).

I installed and built the `xvcd` and it successfully detects the FT232 device similar as I already did for a PicoEVB built-in FT232 device. However, when I connect through Vivado on a remote machine, it tells me that it doesn't detect any Xilinx device within the port (given by `xvcd`).

I would like to ask if you have an example for FT232R wiring to the JTAG port, or in the case, suggest a device to program the Litefury. I connected the Litefury to the PCIe port for NVMM in a Xavier NX, where I am also connecting the FTDI device.

Thanks in advance

RHSResearchLLC, 2020-09-03:
Hello,
Unfortunately, I don't have any experience with the FT232R. I use FT230X on PicoEVB, and I've used FT232H with success (but that code uses the MPSSE, not bitbang). I'm sorry I can't be more help.

For programming I usually use Digilent HS1/HS2/HS3. Can usually find them at a decent price

closed 2020-09-03 by RHSResearchLLC

lleon95, 2020-09-03:
Hi. I just wanted to mention I managed to solve the issue. I used this pinout:

```
      -FTDI-  -JTAG-
    -  RXD  -  TDI
    -  TXD  -  TCK
    -  RTS  -  TDO
    -  CTS  -  TMS
    -  DTR  -  TRST
    -  DCD  -  SRST 
```

With that, I was able to use the XVCD to configure the Litefury.

Regards,
Leon.

RHSResearchLLC, 2020-09-03:
Awesome thanks for the info!