RHSResearchLLC/NiteFury-and-LiteFury #9: Insufficient mounting stability of 6-pin EZmate connectors

eqvinox, 2020-09-02:
Hiya,

I just managed to tear one of the 6-pin EZmate connectors off the small JTAG adapter boards. Details:

- didn't really use force, used to dealing with small flimsy hardware
- the connector only came off about halfway; both of the side retention pins plus 3 or 4 of the data pins
- the pads were *not* ripped off, it seems the solder connection itself was the weak point
- I successfully soldered it back on manually, but burned some plastic on the JTAG connector since the positioning is awkward

I checked my other adapter (I ordered 2 boards) and at first it seemed to be better, but then on that board also one of the side retention pins came off in the same way, and after this happened the connector doesn't unplug properly anymore (lifts the socket instead.) It seems that as soon as one of the side pins is gone, you're kinda f*cked and the entire thing cascade-fails.

I see 3 possible causes here:

- the solder pads or mask are too small so the solder connection is too weak
- the connectors fundamentally are badly designed with a very small area of force transfer
- just bad luck, either on the boards or on my use of them

I'm posting this here as a reference/warning and collection point in case other people have the same problem. I haven't tried/checked this on the actual FPGA boards since I'd rather not break them. Presumably the footprint is the same there, though maybe not the assembly procedure?

Either way it might be a good idea to double check mounting details on these connectors and maybe increase the size of the pads and solder mask opening for the mechanical/mounting pins at the side.

Cheers!

RHSResearchLLC, 2020-09-03:
Hello Hiya,

The issue here is that the first batch (~50 pcs) of the adapter boards were made using low-temp bismuth solder, to prevent the 14-pin connector from melting. Unfortunately, it seems this solder can shear with low force. Bismuth solder was chosen because the assembly has to be lead free for EU, and the 14-pin connector's plastic body would melt at the temperatures required for "standard" (SAC305) lead-free solder.

All subsequent batches use a better 14-pin connector and SAC305 solder, and the solder joint is very strong. The LiteFury/NiteFury itself has always used SAC305 solder so there is no danger of ripping off the connector.

To anyone who experiences this issue, email dave@rhsresearch.com with your address and I'll send you a replacement adapter for free.

eqvinox, 2020-09-07:
> and the 14-pin connector's plastic body would melt at the temperatures required

Can confirm %-)

> The LiteFury/NiteFury itself has always used SAC305 solder so there is no danger of ripping off the connector.

Awesome, very glad to hear that!  That was my primary concern.

> To anyone who experiences this issue, email dave@rhsresearch.com with your address and I'll send you a replacement adapter for free.

Great!  Funnily enough I don't even need the adapter (nothing else I have has 14-pin 2mm JTAG), I was kinda just playing around with it to see how the 6-pin connector works.  So that's 2 pcs that don't need to be replaced.

closed 2023-10-17 by RHSResearchLLC