RHSResearchLLC/NiteFury-and-LiteFury #12: Fan noise

Mcilie, 2021-04-01:
Please, as awesome as this product is, being powerful and high bandwidth for such a low price, is there a way to make the fan quieter?

RHSResearchLLC, 2021-04-04:
Hello,

Unfortunately, there isn't. The fan is hard-wired, and I have a lot of fans in stock. That said, some fans are noisier than others. Generally I've found "good" fans to be pretty quiet. Maybe you have a bad fan?

If you bought from Amazon, you can try returning for a new one. Or, if you are in the US, send me your email and I can send you a new fan. info@rhsresearch.com.

Mcilie, 2021-04-09:
Its Okay, I decided to keep the unit i have and frankenstein my own fan on to there. The original fan was also going against the airflow direction of my PC, so i guess this is the optimal solution.
Thank you though

suoto, 2021-10-04:
Could you provide a PN for the fan(s) you used so people could source their own? I found a good fan, same size but it's 5V but I don't have noise values, air flow values to compare.

RHSResearchLLC, 2021-10-04:
Hello Andre,

Unfortunately I don't have a part number; it was a custom design.

suoto, 2021-10-04:
Do you think this one here: https://noctua.at/en/nf-a4x10-5v/specification would mechanically fit?

closed 2022-08-12 by RHSResearchLLC

gniticxe, 2023-05-01:
I am considering designing an adapter / duct for a replacement fan on the Nitefury.  Mine is mounted on a PCIe expansion card for an otherwise silent desktop.  The Noctua NF-A4x20 5V PWM is a candidate, but would take 2 PCIe slots.  Is there interest in the community for trying my eventual design, or other bits of advice before I begin the design and post to printables.com?