RHSResearchLLC/NiteFury-and-LiteFury #13: Windows development support

c0ffymachyne, 2021-05-06:
From the files found in the repository I am guessing only Linux is supported, is this correct ?

AEW2015, 2021-05-06:
I have used these boards with Windows using the Xilinx XDMA driver or the Jungo PCIe driver (trail version).  It is possible.

RHSResearchLLC, 2021-05-06:
I'm using the XDMA IP provided by Xilinx, and their drivers; so I don't provide any additional support other than what Xilinx offers. That said, Xilinx does seem to have some support for Windows: https://forums.xilinx.com/t5/PCIe-and-CPM/XDMA-Driver-for-Windows-10/td-p/777694

AEW2015, 2021-05-06:
This is my video shows how to use the Xilinx XDMA drivers on Windows. 
https://youtu.be/m56rBYnmxME

c0ffymachyne, 2021-05-06:
Thank you so much ! I will definitely check all those options !

c0ffymachyne, 2021-05-14:
Any chance to have dma-test-2.py script working on windows ?