RHSResearchLLC/NiteFury-and-LiteFury #14: Picoez to JTAG adapter info

neon5d, 2021-06-20:
Hello,

Can you give more detailed information on what connector types are used for the pico (6 pins) to JTAG (14 pins, 2x7) adapter?
Thank you

RHSResearchLLC, 2021-06-21:
Hello,

Here are the design files (schematic, bill of materials, and gerbers):

[Pico-JTAG-BOM.xlsx](https://github.com/RHSResearchLLC/NiteFury-and-LiteFury/files/6689318/Pico-JTAG-BOM.xlsx)
[Pico-JTAG.pdf](https://github.com/RHSResearchLLC/NiteFury-and-LiteFury/files/6689320/Pico-JTAG.pdf)
[Pico-JTAG-Gerber-X2.zip](https://github.com/RHSResearchLLC/NiteFury-and-LiteFury/files/6689321/Pico-JTAG-Gerber-X2.zip)

RHSResearchLLC, 2021-06-21:
Here is the schematic for the 6-pin Pico-ezmate cable as well

[pjc-1.pdf](https://github.com/RHSResearchLLC/NiteFury-and-LiteFury/files/6689327/pjc-1.pdf)

neon5d, 2021-06-21:
These are what I needed. Thank you.

closed 2021-06-21 by neon5d