RHSResearchLLC/NiteFury-and-LiteFury #17: Flashing/program via PCIe

dannycoin, 2021-08-17:
Hi there..

1) can an acorn 215+ / Litefury be flashed through PCIe without JTAG cable? 
2) is it mandatory to use a JTAG cable to flash/program? 
3) Is there a guide to flashing via PCIe?
3) Can I design in tools like Vivado and then flash / program via PCIe?

RHSResearchLLC, 2021-08-17:
1. Yes, but only if there is already a bitstream that supports this. See https://github.com/RHSResearchLLC/NiteFury-and-LiteFury/tree/master/spi-loader
2. Only if you don't already have a compatible bitstream running
3. See https://github.com/RHSResearchLLC/NiteFury-and-LiteFury/tree/master/spi-loader
4. You can use Vivado to create .mcs, then load that MCS using the spi-loader

The factory 215+ bitstream probably won't work with the spi-loader (not sure). But the factory NiteFury bitstream should.

Finally, you can also use a FTDI for programming, with OpenOCD. Vivado doesn't recognize FTDI but OpenOCD does. I don't have an example for NiteFury, but here is an example for PicoEVB, which is similar: https://github.com/RHSResearchLLC/PicoEVB/tree/master/spi-flash-program-openocd

Gbps, 2022-10-06:
I wrote up a guide here using the above info, hopefully this can help:
https://github.com/Gbps/nitefury-openocd-flashing-guide

RHSResearchLLC, 2022-10-06:
Nice! I tried this same exact thing but it didn't seem to work, and I gave up rather quickly. Now that I think about it, I was trying on a blank flash, which probably wasn't properly configured. *Furys that have been programmed once have the flash preconfigured. Maybe I'll integrate your work with that caveat. Thanks again for this nice writeup!