RHSResearchLLC/NiteFury-and-LiteFury #19: Early processing property not working in Vivado Linux

VHWeng, 2022-10-07:
Basically Vivado in Windows works fine. The problem is using Vivado in Linux. The PCIe pins are miss-placed.  Vivado reports an error in synthesis: _[Project 1-236] Implementation specific constraints were found while reading constraint file [ …/constraints/early.xdc]. These constraints will be ignored for synthesis but will be used in implementation. Impacted constraints are listed in the file [.Xil/Top_wrapper_propImpl.xdc]._ 
To fix this problem, I rebuild the build script in Linux to resets the LCO property on the PCIe cell and add the correct LCO. A couple of points, the synthesis needs to be opened to make the change. But you can’t close it. Closing it will lose the change or reset the synthesis. Close the design after implementation (see attached) . 

I'm using the Vivado 2022.1 Any idea? 

VH

[build_mod.zip](https://github.com/RHSResearchLLC/NiteFury-and-LiteFury/files/9729745/build_mod.zip)

RHSResearchLLC, 2022-10-07:
If it works different on Linux than Windows, its a Vivado issue. I can't remember if I used Linux or Windows to build the factory bitstreams. At any rate, there are "Early", constraints which are processed first, and those should be pinning the location of the transceiver BEL. Make sure the properties for the early constraints file has the processing order set to "early"
[image]

VHWeng, 2022-10-08:
Yes, this seems like it’s a Xilinx Vivado issue. I posted this mostly to see if anyone else was having the same problem. I recheck the early processing order, and it is set to early. I’m using Ubuntu 20.04 (_Linux XLinux 5.15.0-48-generic #54~20.04.1-Ubuntu SMP Thu Sep 1 16:17:26 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux)_ with Vivado 2021.1.

Bobo1239, 2022-10-29:
I've also noticed this issue with Vivado 2021.1 running on (Arch) Linux. My workaround was to set the processing order of `Top_xdma_0_0_pcie2_ip-PCIE_X0Y0.xdc` to  NORMAL. (not sure if this causes any other issues). But indeed seems like a Vivado bug and should be reported to XiIinx.

quizitlive, 2022-12-19:
I've had the issue with Vivado 2022.2 on Ubuntu 22.04.1 -- I had to change the `Top_xdma_0_0_pcie2_ip-PCIE_X0Y0.xdc to NORMAL too. However, as well as the expected 3 critical warnings, I get another, that it failed to meet timing violations -- related to Inter-Clock Path Setup userclk2 to mmcm_clkout0  (related to he CodeBlinker). Is this related to other contraints being applied in the wrong order?

RHSResearchLLC, 2022-12-19:
@quizitlive no the timing violation warning for codeblinker is unrelated.

teknoman117, 2024-02-05:
> I've also noticed this issue with Vivado 2022.1 running on (Arch) Linux. My workaround was to set the processing order of `Top_xdma_0_0_pcie2_ip-PCIE_X0Y0.xdc` to NORMAL. (not sure if this causes any other issues). But indeed seems like a Vivado bug and should be reported to XiIinx.
> 
> (Just installed 2022.2 today so will also try a build with that in case it was already fixed.)
> 
> Update: Not fixed in 2022.2 afaict.

I have the same issue running Vivado 2022.1 and 2023.2 on Gentoo Linux.

If anyone stumbles across this issue and was lost on how exactly to make that change:

1. Run the "rebuild.tcl" script (or run synthesis & implementation). The constraints file won't be visible until that is run
2. Go to the "Sources" pane
3. Click the "IP Sources" tab
4. Hit the search button (top left of pane)
5. Type "Top_xdma_0_0_pcie2_ip-PCIE_X0Y0"
6. Select the xdc file that comes up
7. Go to the "Source File Properties" pane
8. Click the "Properties" tab
9. Change the "PROCESSING_ORDER" property to "NORMAL"
10. Run the "build.tcl" script (or run synthesis & implementation again)

[image]