RHSResearchLLC/NiteFury-and-LiteFury #20: lspci -vvv -s 01:00.0 displays  "Unknown Header Type 7F"

bipingamiyv, 2023-01-24:
Hi, I am using NITEFURY 2(Artix-7 xc7a200tfbg484-2) and created the design with XDMA IP. I am programming the board through JTAG only.   

1.  I loaded the bin file provided by RHSResearchLLC(https://github.com/RHSResearchLLC/NiteFury-and-LiteFury/tree/master/Sample-Projects/Project-0/FPGA/Nitefury-II/mcs) through JTAG it worked. Device was detecting in pcie.
2. Then i generate the bitfile and loaded it thorough JTAG. lspci -vvv -s 01:00.0 displays "Unknown Header Type 7F". I tried this command echo 1 > /sys/bus/pci/rescan also, but still lspci -vvv -s 01:00.0 displays "Unknown Header Type 7F"
3. Again i loaded their original bin file. It worked. Device was detecting in pcie bus.
4. Then i created the binfile (using my bitfile) by following (write_cfgmem -format bin -size 16 -interface SPIx4 -force -loadbit "up 0 ./project.runs/impl_1/Top_wrapper.bit" -file "../mcs/out.bin")  and loaded it to board through JTAG. Again lspci -vvv -s 01:00.0 displays "Unknown Header Type 7F". I tried this command  echo 1 > /sys/bus/pci/rescan also, but still lspci -vvv -s 01:00.0 displays "Unknown Header Type 7F"

I am using the same XDMA IP configuration as provided in sample project,.

These are the questions I have,

1.     Is the binfile packaging I am doing (write_cfgmem -format bin -size 16 -interface SPIx4 -force -loadbit "up 0 ./project.runs/impl_1/Top_wrapper.bit" -file "../mcs/out.bin") is correct ?
2. Not sure why loading bitfile via JTAG also doesnt work ? Is there any linux command I need to run to re-enumerate the PCI device ?

RHSResearchLLC, 2023-01-24:
That "Unknown Header Type..." error usually occurs when the Nitefury is enumerated on the bus (at host boot time) but then drops off and comes back. This will happen when you reload bitfile or reprogram flash via JTAG.

The solution is to re-enumerate the bus. However, in practice, I've never seen this work. I've tried dozens of systems and have yet to see `echo 1 > /sys/bus/pci/rescan` work. 

Sorry I can't be more helpful.

bipingamiyv, 2023-01-31:
Thanks for your response.

We packaged the bitfile using the script and loaded via JTAG.

[image]

Is the binfile packaging I am doing (write_cfgmem -format bin -size 16 -interface SPIx4 -force -loadbit "up 0 ./project.runs/impl_1/Top_wrapper.bit" -file "../mcs/out.bin") is correct ?

bipingamiyv, 2023-02-07:
It's done.

closed 2023-02-07 by bipingamiyv

mayureshw, 2023-10-20:
I came across "Unknown Header Type 7F".

A basic question: When you see "Unknown Header Type 7F" does the design work? i.e. is this just a cosmetic issue that can be ignored or it leads to any problem with the driver?

bipingamiyv, 2023-10-22:
"Vivado initially sets the processing order of 'Top_xdma_0_0_pcie2_ip-PCIE_X0Y0.xdc' to 'EARLY' by default. However, I changed the order to  'NORMAL,' which resolved the issue."

You can reffer to the below link also:
[https://github.com/RHSResearchLLC/NiteFury-and-LiteFury/issues/19](url)