RHSResearchLLC/NiteFury-and-LiteFury #23: Where to find/how to change "board version" (aka `board_pvrsn`)?

mablinov, 2023-03-03:
Probably a stupid question, but I can't figure out where to find the right value to put for the "board version" parameter that is required for the [`furytest.sh`](https://github.com/RHSResearchLLC/NiteFury-and-LiteFury/blob/master/Sample-Projects/Project-0/Host/fury-test.sh) script.

In the `[test-general.py](https://github.com/RHSResearchLLC/NiteFury-and-LiteFury/blob/9fc9f1bae6c06523d46470fc309b73e7015aa9fd/Sample-Projects/Project-0/Host/test-general.py#L50)` its referred to as the `board_vrsn` parameter. I added a line to print out the `pvrsn` read back from the board, and it appears to be assigned to `2` - Where is this configured from the Vivado side of things?

RHSResearchLLC, 2023-03-03:
[image]

RHSResearchLLC, 2023-03-03:
[image]

mablinov, 2023-03-03:
Got it, thanks!

closed 2023-03-03 by mablinov