RHSResearchLLC/NiteFury-and-LiteFury #25: Timing Violations with LiteFury on Vivado 2022.1

dwilliamson, 2023-04-17:
With 2022.1 on Windows and the latest version of this repo, I am getting the 3 expected critical warnings, as well as some timing violations:

[image]

These are from Inter-Clock Paths `userclk2` to `mmcm_clkout0` (GPIO to CodeBlinker):

[image]
[image]

I have updated to this specific version as you mentioned at https://github.com/RHSResearchLLC/NiteFury-and-LiteFury/issues/15#issuecomment-1213275575 that the release is supported. Are you seeing this as well?

Thanks,
- Don

RHSResearchLLC, 2023-04-24:
I haven't tried 2022.1. I don't see any timing violations when I build it.

dwilliamson, 2023-04-26:
How can you not see any timing violations when you build it if you haven't tried 2022.1? If it requires updating to 2022.2 then I'm happy with that. This also happens with a clean build on the NiteFury project.

Anyway, our currently supported Vivado version is 2019.1 so I simply went back 2 or 3 years in this repo to get everything working for me.

Thanks,
- Don

closed 2023-04-26 by dwilliamson

Chester-Gillon, 2023-05-29:
> With 2022.1 on Windows and the latest version of this repo, I am getting the 3 expected critical warnings, as well as some timing violations:
> These are from Inter-Clock Paths `userclk2` to `mmcm_clkout0` (GPIO to CodeBlinker):

Just to say I get the same timing violations on the same Inter-Clock Paths `userclk2` to `mmcm_clkout0` paths when built the latest version of **NiteFury** with 2022.1 on Windows.  I haven't yet investigated the significance of the timing violations, but after loading the built bitstream over JTAG (i.e. left the flash unchanged) basic functionality worked:
1. Read the NiteFury identity from the GPIO registers.
2. A test of the DDR passed.
3. Read the temperature and voltages from the XADC.

Chester-Gillon, 2023-06-04:
> With 2022.1 on Windows and the latest version of this repo, I am getting the 3 expected critical warnings, as well as some timing violations:
> These are from Inter-Clock Paths `userclk2` to `mmcm_clkout0` (GPIO to CodeBlinker):

Looking at the _Methodology_  tab in Vivado there is critical warning which contains ``TIMING #2 Critical Warning The clocks userclk2 and mmcm_clkout0 are related (timed together) but they have no common primary clock. The design could fail in hardware. To find a timing path between these clocks, run the following command: report_timing -from [get_clocks userclk2] -to [get_clocks mmcm_clkout0] ``

Running the suggested command in the TCL Console results in:
```
report_timing -from [get_clocks userclk2] -to [get_clocks mmcm_clkout0]
INFO: [Timing 38-91] UpdateTimingParams: Speed grade: -2, Delay Type: max.
INFO: [Timing 38-191] Multithreading enabled for timing update using a maximum of 2 CPUs
INFO: [Timing 38-78] ReportTimingParams: -from_pins  -to_pins  -max_paths 1 -nworst 1 -delay_type max -sort_by slack.
WARNING: [Timing 38-164] This design has multiple clocks. Inter clock paths are considered valid unless explicitly excluded by timing constraints such as set_clock_groups or set_false_path.
Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
-----------------------------------------------------------------------------------------
| Tool Version : Vivado v.2022.1 (win64) Build 3526262 Mon Apr 18 15:48:16 MDT 2022
| Date         : Sun Jun  4 13:47:05 2023
| Host         : DESKTOP-EG5EBI0 running 64-bit major release  (build 9200)
| Command      : report_timing -from [get_clocks userclk2] -to [get_clocks mmcm_clkout0]
| Design       : Top_wrapper
| Device       : 7a200t-fbg484
| Speed File   : -2  PRODUCTION 1.23 2018-06-13
-----------------------------------------------------------------------------------------

Timing Report

Slack (VIOLATED) :        -2.210ns  (required time - arrival time)
  Source:                 Top_i/GPIO_regs/axi_gpio_2/U0/gpio_core_1/Dual.gpio2_Data_Out_reg[27]/C
                            (rising edge-triggered cell FDRE clocked by userclk2  {rise@0.000ns fall@4.000ns period=8.000ns})
  Destination:            Top_i/CodeBlinker_3/inst/s_code_reg[2]/D
                            (falling edge-triggered cell FDRE clocked by mmcm_clkout0  {rise@0.000ns fall@5.000ns period=10.000ns})
  Path Group:             mmcm_clkout0
  Path Type:              Setup (Max at Slow Process Corner)
  Requirement:            1.000ns  (mmcm_clkout0 fall@25.000ns - userclk2 rise@24.000ns)
  Data Path Delay:        0.641ns  (logic 0.379ns (59.094%)  route 0.262ns (40.906%))
  Logic Levels:           0  
  Clock Path Skew:        -2.257ns (DCD - SCD + CPR)
    Destination Clock Delay (DCD):    3.738ns = ( 28.738 - 25.000 ) 
    Source Clock Delay      (SCD):    5.995ns = ( 29.995 - 24.000 ) 
    Clock Pessimism Removal (CPR):    0.000ns
  Clock Uncertainty:      0.289ns  ((TSJ^2 + DJ^2)^1/2) / 2 + PE
    Total System Jitter     (TSJ):    0.071ns
    Discrete Jitter          (DJ):    0.266ns
    Phase Error              (PE):    0.151ns
  Clock Domain Crossing:  Inter clock paths are considered valid unless explicitly excluded by timing constraints such as set_clock_groups or set_false_path.

    Location             Delay type                Incr(ns)  Path(ns)    Netlist Resource(s)
  -------------------------------------------------------------------    -------------------
                         (clock userclk2 rise edge)
                                                     24.000    24.000 r  
    GTPE2_CHANNEL_X0Y6   GTPE2_CHANNEL                0.000    24.000 r  Top_i/xdma_0/inst/Top_xdma_0_0_pcie2_to_pcie3_wrapper_i/pcie2_ip_i/inst/inst/gt_top_i/pipe_wrapper_i/pipe_lane[0].gt_wrapper_i/gtp_channel.gtpe2_channel_i/TXOUTCLK
                         net (fo=1, routed)           1.274    25.274    Top_i/xdma_0/inst/Top_xdma_0_0_pcie2_to_pcie3_wrapper_i/pcie2_ip_i/inst/inst/gt_top_i/pipe_wrapper_i/pipe_clock_int.pipe_clock_i/txoutclk_i.txoutclk_i_0
    BUFGCTRL_X0Y20       BUFG (Prop_bufg_I_O)         0.081    25.355 r  Top_i/xdma_0/inst/Top_xdma_0_0_pcie2_to_pcie3_wrapper_i/pcie2_ip_i/inst/inst/gt_top_i/pipe_wrapper_i/pipe_clock_int.pipe_clock_i/txoutclk_i.txoutclk_i/O
                         net (fo=1, routed)           1.520    26.874    Top_i/xdma_0/inst/Top_xdma_0_0_pcie2_to_pcie3_wrapper_i/pcie2_ip_i/inst/inst/gt_top_i/pipe_wrapper_i/pipe_clock_int.pipe_clock_i/refclk
    MMCME2_ADV_X0Y2      MMCME2_ADV (Prop_mmcme2_adv_CLKIN1_CLKOUT3)
                                                      0.077    26.951 r  Top_i/xdma_0/inst/Top_xdma_0_0_pcie2_to_pcie3_wrapper_i/pcie2_ip_i/inst/inst/gt_top_i/pipe_wrapper_i/pipe_clock_int.pipe_clock_i/mmcm_i/CLKOUT3
                         net (fo=1, routed)           1.559    28.510    Top_i/xdma_0/inst/Top_xdma_0_0_pcie2_to_pcie3_wrapper_i/pcie2_ip_i/inst/inst/gt_top_i/pipe_wrapper_i/pipe_clock_int.pipe_clock_i/userclk2
    BUFGCTRL_X0Y0        BUFG (Prop_bufg_I_O)         0.081    28.591 r  Top_i/xdma_0/inst/Top_xdma_0_0_pcie2_to_pcie3_wrapper_i/pcie2_ip_i/inst/inst/gt_top_i/pipe_wrapper_i/pipe_clock_int.pipe_clock_i/userclk2_i1.usrclk2_i1/O
                         net (fo=22601, routed)       1.404    29.995    Top_i/GPIO_regs/axi_gpio_2/U0/gpio_core_1/s_axi_aclk
    SLICE_X93Y139        FDRE                                         r  Top_i/GPIO_regs/axi_gpio_2/U0/gpio_core_1/Dual.gpio2_Data_Out_reg[27]/C
[... cut ...]
  -------------------------------------------------------------------
                         required time                         28.426    
                         arrival time                         -30.636    
  -------------------------------------------------------------------
                         slack                                 -2.210    

report_timing: Time (s): cpu = 00:00:25 ; elapsed = 00:00:12 . Memory (MB): peak = 4222.516 ; gain = 53.934
```
Haven't yet determined what to change to fix the timing violations.