RHSResearchLLC/NiteFury-and-LiteFury #28: Would Nitefurry fit in Intel NUC 13?

mayureshw, 2023-09-04:
I have ordered Nitefurry, but yet to order a suitable SBC / PC with M.2 M key slot.

Considering NUC-13 i7. It has a M.2 M key slot, but not sure about whether it will accommodate the height, considering the heat sink.

Please suggest alternative SBCs otherwise.

mayureshw, 2023-09-04:
Please confirm, the given dimension, 22x80x28mm, is inclusive of heat sink. Right?

RHSResearchLLC, 2023-09-04:
Yes those dimensions include the heatsink. I don't know if it fits in a NUC-13. Let me know if it does. Thanks

mayureshw, 2023-09-11:
Nope :-(

It produces out by about 5mm or so. (Didn't actually measure with a scale.)

Also removed, the Sata casing on the lid but that doesn't help.

But I am still going to use it by keeping the bottom lid open, keeping the NUC upside down! Barring the fitting issue it seems a good combination - NUC + Nitefurry.

closed 2023-09-11 by mayureshw