RHSResearchLLC/NiteFury-and-LiteFury #32: Which partname to set for NiteFury: XC7A200T-2FBG484E or XC7A200T-FBG484

mayureshw, 2023-09-21:
Readme of this repository shows XC7A200T-2FBG484E, but Vivado rejects this as invalid partname when creating bitstream. (I am using all tcl scripts, no UI.)

In the following xdc file in this repository I found partname XC7A200T-FBG484
Sample-Projects/Project-0/FPGA/Nitefury-II/project/project.srcs/sources_1/bd/Top/ip/Top_mig_7series_0_0/Top_mig_7series_0_0/example_design/par/example_top.xdc

This is accepted by Vivado, but a simple design to just switch on or off an LED is transferred to the expected effect on LED doesn't happen.

mayureshw, 2023-09-21:
Ah! '0' means ON for LED on this board. I can now control LED_A*

But what about LED_M2, it doesn't seem to matter whether I set it to '1' or '0'.

RHSResearchLLC, 2023-10-17:
xc7a200tfbg484-2 is the right part.

The LED_M2 is not on the board- it toggles the LED line on the M.2 connector, which usually isn't connected to anything. Some motherboards will pin this to a header or LED.

closed 2023-10-17 by RHSResearchLLC