RHSResearchLLC/NiteFury-and-LiteFury #36: What should be the constraints on sys_clk_clk_p and _n

mayureshw, 2023-10-19:
I see IOSTANDARD property for sys_clk_clk_p and _n in normal.xdc

What are the PACKAGE_PIN and create_clock constraints needed on these ports?

The comment in the xdc suggests, MIG might be setting those. But I am not using MIG in my design.

```
###############################################################################
# DDR
###############################################################################
# Note: Most of the pins are set in the constraints file created by MIG
set_property IOSTANDARD LVDS_25 [get_ports sys_clk_clk_p]
set_property IOSTANDARD LVDS_25 [get_ports sys_clk_clk_n]
```

vogma, 2023-11-10:
Hi @mayureshw,

correct, the PACKAGE_PIN and create_clock constraints are set by the MIG Constraint File.
You can find this file in the Vivado GUI under `Sources -> IP Sources Tab -> Synthesis -> Top_mig_7series_0_0 -> Top_mig_7series_0_0.xdc` or in the project structure under `project.gen/sources_1/bd/Top/ip/Top_mig_7series_0_0_1/Top_mig_7series_0_0/user_design/constraints`

Hope this helps.
Marco

mayureshw, 2023-11-10:
Ok

closed 2023-11-10 by mayureshw