RHSResearchLLC/NiteFury-and-LiteFury #42: NiteFury II / Ubuntu 22.04 / Envoy Express Thunderbolt - Success

timkpaine, 2023-12-09:
Just wanted to post for others who might be looking, pretty successful with the following devices (taking note of https://github.com/RHSResearchLLC/NiteFury-and-LiteFury/issues/19 when you want to make changes).

- Fine on [m.2 nvme adapter](https://www.amazon.com/gp/product/B07JJTVGZM) / vanilla xilinx xdma drivers
- Fine on [a standard thundebolt nvme enclosure](https://eshop.macsales.com/shop/envoy-express/thunderbolt-3?utm_term=&utm_campaign=&utm_source=google&utm_medium=cpc&hsa_acc=9360794409&hsa_cam=20395132005&hsa_grp=&hsa_ad=&hsa_src=x&hsa_tgt=&hsa_kw=&hsa_mt=&hsa_net=adwords&hsa_ver=3&gad_source=1&gclid=EAIaIQobChMIn_iksZaBgwMVw2NHAR1KPwQsEAAYASAAEgLDEfD_BwE) (but note that this might be motherboard dependent, only one of my two thunderbolt ports will work on [my motherboard](https://www.gigabyte.com/Motherboard/GA-Z170X-UD5-TH-rev-10#ov)

I will test soon if it works on an M1 Macbook passthrough to Ubuntu via UTM.

closed 2023-12-09 by timkpaine

teknoman117, 2024-02-05:
This enclosure also worked for me (Both a LiteFury and an Acorn CLE-215+)

https://www.amazon.com/Enclosure-Aluminum-External-Support-Compatible/dp/B0BBZT42HC

vrbadev, 2024-11-10:
I have recently bought the Nitefury board from Amazon and I am currently testing it with [this Hagibis NVMe enclosure](https://www.shophagibis.com/products/usb4-m-2-nvme-ssd-enclosure-with-led-display-hb777). It contains ASM2464PD chip which should be capable of PCIe Gen4 x4. So I thought it could be used the same way as the usually recommended enclosures with Intel chipsets JHL6340, JHL7540, JHL7440 etc. 
The thing is that when I connect it, I do not see any change in `lsusb` nor `lspci` output. In output of `dmesg` I get only `ucsi_acpi USBC000:00: UCSI_GET_PDOS failed (-95)` which is probably related to failing power delivery negotiation. Could anybody confirm what the expected outputs are, and if the enclosure I am testing is compatible? 
Also, the enclosure reaches approx. 80˚C quickly even with the default bitstream flashed - is that normal? My Digilent HS2 JTAG programmer sometimes keeps reconnecting every cca 1s in the Vivado Hardware Manager with this message: `HW Target shutdown. Closing target: localhost:3121/xilinx_tcf/Digilent/210249BAC3B8` - can it be caused by some temperature protection or is the board/programmer just faulty?