RHSResearchLLC/NiteFury-and-LiteFury #44: How to handle constraints files?

Gandalf1783, 2024-01-02:
Hey!

I got the Litefury model and during designing my VHDL code I remebered the constrains file.

AFAIK, the constraints are used to set the physical constraints of the FPGA.
However, can I just drop them into ANY project?

Or do I have to refactor some properties in there?

In my case, I did not take a look onto the default projects.

**TLDR**: Are the constraints files (early/normal.xdc) project-dependent, a "must-have" in every project or not necessary at all when going to generate and flash the bitstream onto the FPGA?

Thanks!

RHSResearchLLC, 2024-01-03:
Hello,

In general constraint files are necessary for every project. Xilinx lets you put all sorts of stuff in constaints files- like physical constraints, timing constraints, and settings for properties like bitstream generation.

"Early" constraint files take effect before "Normal" and "Late". In this project, the early constraints are necessary because the XDMA IP tries to locate the transceivers in a certain spot, which doesn't match up with the wiring on the hardware. And once they are loc'd, the tool won't re-loc when it comes across a new constraint for the same entity. So the early constraint file "locks in" the location of the transceivers for PCIe.

Gandalf1783, 2024-01-04:
Ahh that clears some things up!

Thanks!

closed 2024-01-04 by Gandalf1783