RHSResearchLLC/NiteFury-and-LiteFury #55: Device not showing up under lspci

LowEnergeticAg, 2024-04-17:
I have a NiteFury II, but it's only detected by lspci with the bitstream it came with and the one in the repo. When I tried to get my own project working with this configuration [image]
[image]
[image] it wouldn't show up under lspci. I also tried synthesizing the project from this repo myself, but this also didin't work.

RHSResearchLLC, 2024-04-18:
1. Make sure the constraints are taking effect (and make sure you are using the constraints from the depo. This is a common problem. There are other github issues that cover this, but basically under some circumstances Vivado doesn't respect the constraints- the pinout for PCIe. Look at your design schematic view in Vivado and make sure the PCIe pins out correctly.

See here for the constraints: https://github.com/RHSResearchLLC/NiteFury-and-LiteFury/tree/master/Sample-Projects/Project-0/FPGA/Nitefury-II/project/project.srcs/constrs_1/imports/constraints

Make sure early.xdc is setup to as "early" in Vivado- this is a separate step that must be taken. Otherwise Vivado uses the default constraints which aren't correct for this design. Another option is to add a 'reset' to the pinout before these early constraints, check out other github issues in this repo for details.

3. Are you loading your bitstream into flash? PCIe device needs to be ready within 100ms of powerup of the host. So programming your bitstream into fabric after the system is up, usually doesn't work. Your bitstream need to be in flash, or alternatively if you have a way to reset the host (rescan doesn't work- I mean a true full reset) without dropping power on the M.2 slot, that also works.

LowEnergeticAg, 2024-04-18:
Thanks for the reply. PCIe pins where in fact not mapped correctly and I was able to resolve the issue by setting the compile order for the constraints file with the PCIe pinout to early as you suggested.

closed 2024-04-18 by LowEnergeticAg