RHSResearchLLC/NiteFury-and-LiteFury #56: DRC errors with Vivao 2023.2

gonsolo, 2024-04-22:
I get the following errors using Vivado 2023.2 on Linux running build.tcl:

> ERROR: [DRC NSTD-1] Unspecified I/O Standard: 6 out of 6 logical ports use I/O standard (IOSTANDARD) value 'DEFAULT', instead of a user assigned specific value. This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all I/O standards. This design will fail to generate a bitstream unless all logical ports have a user specified I/O standard value defined. To allow bitstream creation with unspecified I/O standard values (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks NSTD-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run. Problem ports: code[2:0], clk, led, and ok.

> ERROR: [DRC UCIO-1] Unconstrained Logical Port: 6 out of 6 logical ports have no user assigned specific location constraint (LOC). This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all pin locations. This design will fail to generate a bitstream unless all logical ports have a user specified site LOC constraint defined.  To allow bitstream creation with unspecified pin locations (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks UCIO-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run.  Problem ports: code[2:0], clk, led, and ok. 

Any advice?

makslevental, 2024-04-29:
Did you manage to figure this out?

gonsolo, 2024-04-30:
Yes. I manually fixed the ports (clk, led, ok) in Vivado.

closed 2024-04-30 by gonsolo

makslevental, 2024-04-30:
> Yes. I manually fixed the ports (clk, led, ok) in Vivado.

Which `IOSTANDARD` did you set?

gonsolo, 2024-04-30:
Ha, hard to remember after one week. :roll_eyes: 
I just played around until it worked but I *believe* it was just a matter of setting it explicitly to the default. I have no idea why that would make a difference.