RHSResearchLLC/NiteFury-and-LiteFury #60: Out of stock

danielzgtg, 2024-10-29:
It says the Nitefury is "SOLD OUT" since [April](https://web.archive.org/web/20240412133709/https://rhsresearch.com/collections/rhs-public/products/nitefury-xilinx-artix-fpga-kit-in-nvme-ssd-form-factor-2280-key-m) and [June](https://web.archive.org/web/20240620210110/https://rhsresearch.com/collections/rhs-public/products/nitefury-xilinx-artix-fpga-kit-in-nvme-ssd-form-factor-2280-key-m) 2024. When will it be back in stock? For how many years will it be guaranteed to be in production until at least?

RHSResearchLLC, 2024-10-29:
Hello,

There are some on Amazon still, but once those are gone, they will likely
be out of stock forever. I'm not producing any more

danielzgtg, 2024-10-30:
Oh no :(. The other M.2 FPGA on Google is like $500, a 250% increase, overpriced for new grads. NiteFury was the highest LUT count per dollar I could find...

Amazon says "Temporarily out of stock." Based on what you said, I placed an order anyway, hoping it will go through.

closed 2024-10-30 by danielzgtg

RHSResearchLLC, 2024-10-30:
Yeah I got a good deal on FPGAs but it's not worth the outlay to make
another batch. Id probably have to charge $350/ea on the next batch to make
it worthwhile. Even then, these don't sell quickly, so it's a large outlay
and a slooow return.

There are plenty of Litefury in stock, it's not very popular even given the
price. But as a tool to experiment with PCIe on FPGA, Litefury is a good
deal

timkpaine, 2025-03-30:
I've done a bunch of research in this area and I think i might've been the one to purchase the last few (sorry!). Its more expensive, but this board is also good for the price, though the price is much higher: https://www.en.alinx.com/Product/FPGA-Development-Boards/Kintex-7/AX7325B.html

I have not used this board, but it is similar (again, more expensive): https://numato.com/product/aller-artix-7-fpga-board-with-m-2-interface/

Education discount is available (https://numato.com/academic-open-source/#:~:text=We%20do%20offer%20a%20minimum,discounts%20or%20even%20free%20products.).